INTI-CMNB/KiDiff #9: [BUG] KiDiff leaves intermediate png files for PCB, not SCH when no difference, and asked only for differences

oliv3r, 2023-12-18:
Using the kicad 7 full docker image, generating a pcb diff, I am left with all PNG files in the output dir, even when cache_dir is set to blank, set to `/tmp` or even removed entirely. While the png files are great for debugging, once the pdf is generated as final output, they become useless.

```yaml
outputs:
  - name: 'diffsch'
    comment: 'Generates a PDF with the differences between schematic.'
    type: 'diff'
    dir: 'output/diff'
    options: &diff_options
      add_link_id: false
      always_fail_if_missing: true
      cache_dir: '/tmp/pdfdiff'
      copy_instead_of_link: false
      diff_mode: 'red_green'
      force_checkout: false
      fuzz: 5
      new: ''
      # [string='current'] [git,file,output,multivar,current] How to interpret the `new` name. Use `git` for a git hash, branch, etc.
      new_type: 'current'
      old: 'HEAD'
      old_type: 'git'
      only_different: true
      only_first_sch_page: false
      threshold: 0
      use_file_id: false
      pcb: false
    layers: 'all'
  - name: 'diffpcb'
    comment: 'Generates a PDF with the differences between PCBs.'
    type: 'diff'
    dir: 'output/diff'
    options:
      <<: *diff_options
      pcb: true
    layers: 'all'
```

<details>
<summary>-vvv log</summary>
```
- 'Generates a PDF with the differences between PCBs.' (diffpcb) [diff]
DEBUG:Output destination: /workspace/breakout_hdmi/output/diff (kibot - kiplot.py:364)
DEBUG:Starting tool check for KiCad PCB/SCH Diff (kibot - dep_downloader.py:833)
DEBUG:- Checking binary tool KiCad PCB/SCH Diff (kibot - dep_downloader.py:727)
DEBUG:- Looking for tool `kicad-diff.py` at system level (kibot - dep_downloader.py:661)
DEBUG:- Checking version for `/usr/bin/kicad-diff.py` (kibot - dep_downloader.py:628)
DEBUG:- Needed version (2, 4, 4) (kibot - dep_downloader.py:644)
DEBUG:- Cached version (2, 4, 7, 0) (kibot - dep_downloader.py:648)
DEBUG:- Returning `/usr/bin/kicad-diff.py` (kibot - dep_downloader.py:840)
DEBUG:Starting tool check for Git (kibot - dep_downloader.py:833)
DEBUG:- Checking binary tool Git (kibot - dep_downloader.py:727)
DEBUG:- Looking for tool `git` at system level (kibot - dep_downloader.py:661)
DEBUG:- Checking version for `/usr/bin/git` (kibot - dep_downloader.py:628)
DEBUG:- No particular version needed (kibot - dep_downloader.py:642)
DEBUG:- Cached version (2, 39, 2, 0) (kibot - dep_downloader.py:648)
DEBUG:- Returning `/usr/bin/git` (kibot - dep_downloader.py:840)
DEBUG:Including layers: (kibot - out_diff.py:484)
DEBUG:- F.Cu (0) (kibot - out_diff.py:488)
DEBUG:- In1.Cu (1) (kibot - out_diff.py:488)
DEBUG:- In2.Cu (2) (kibot - out_diff.py:488)
DEBUG:- B.Cu (31) (kibot - out_diff.py:488)
DEBUG:- B.Adhesive (32) (kibot - out_diff.py:488)
DEBUG:- F.Adhesive (33) (kibot - out_diff.py:488)
DEBUG:- B.Paste (34) (kibot - out_diff.py:488)
DEBUG:- F.Paste (35) (kibot - out_diff.py:488)
DEBUG:- B.Silkscreen (36) (kibot - out_diff.py:488)
DEBUG:- F.Silkscreen (37) (kibot - out_diff.py:488)
DEBUG:- B.Mask (38) (kibot - out_diff.py:488)
DEBUG:- F.Mask (39) (kibot - out_diff.py:488)
DEBUG:- User.Drawings (40) (kibot - out_diff.py:488)
[... 607 lines cut ...]
DEBUG:kicad-diff.py:Executing: bash -c '( convert "/tmp/pdfdiff/96a4d6308b5290b6e2df51186d5219c8b9352022/55.png" miff:- ;  convert "/tmp/pdfdiff/edbb34bfe69a279a577b4f9071eb005e6b8f84b6/55.png" miff:- ) | convert - \( -clone 0-1 -compose darken -composite \)  -font helvetica -pointsize 30 -draw "text 10,30 '"'"'Layer: User.6'"'"'"  -channel RGB -combine "/workspace/breakout_hdmi/output/diff/diff-550.png"'
DEBUG:kicad-diff.py:Executing: identify -format '%[colorspace]' /workspace/breakout_hdmi/output/diff/diff-550.png
DEBUG:kicad-diff.py:/tmp/pdfdiff/edbb34bfe69a279a577b4f9071eb005e6b8f84b6/56.pdf already converted to PNG
DEBUG:kicad-diff.py:/tmp/pdfdiff/96a4d6308b5290b6e2df51186d5219c8b9352022/56.pdf already converted to PNG
INFO:kicad-diff.py:Creating diff for User.7
DEBUG:kicad-diff.py:Executing: bash -c '( convert "/tmp/pdfdiff/96a4d6308b5290b6e2df51186d5219c8b9352022/56.png" miff:- ;  convert "/tmp/pdfdiff/edbb34bfe69a279a577b4f9071eb005e6b8f84b6/56.png" miff:- ) | convert - \( -clone 0-1 -compose darken -composite \)  -font helvetica -pointsize 30 -draw "text 10,30 '"'"'Layer: User.7'"'"'"  -channel RGB -combine "/workspace/breakout_hdmi/output/diff/diff-560.png"'
DEBUG:kicad-diff.py:Executing: identify -format '%[colorspace]' /workspace/breakout_hdmi/output/diff/diff-560.png
DEBUG:kicad-diff.py:/tmp/pdfdiff/edbb34bfe69a279a577b4f9071eb005e6b8f84b6/57.pdf already converted to PNG
DEBUG:kicad-diff.py:/tmp/pdfdiff/96a4d6308b5290b6e2df51186d5219c8b9352022/57.pdf already converted to PNG
INFO:kicad-diff.py:Creating diff for User.8
DEBUG:kicad-diff.py:Executing: bash -c '( convert "/tmp/pdfdiff/96a4d6308b5290b6e2df51186d5219c8b9352022/57.png" miff:- ;  convert "/tmp/pdfdiff/edbb34bfe69a279a577b4f9071eb005e6b8f84b6/57.png" miff:- ) | convert - \( -clone 0-1 -compose darken -composite \)  -font helvetica -pointsize 30 -draw "text 10,30 '"'"'Layer: User.8'"'"'"  -channel RGB -combine "/workspace/breakout_hdmi/output/diff/diff-570.png"'
DEBUG:kicad-diff.py:Executing: identify -format '%[colorspace]' /workspace/breakout_hdmi/output/diff/diff-570.png
DEBUG:kicad-diff.py:/tmp/pdfdiff/edbb34bfe69a279a577b4f9071eb005e6b8f84b6/58.pdf already converted to PNG
DEBUG:kicad-diff.py:/tmp/pdfdiff/96a4d6308b5290b6e2df51186d5219c8b9352022/58.pdf already converted to PNG
INFO:kicad-diff.py:Creating diff for User.9
DEBUG:kicad-diff.py:Executing: bash -c '( convert "/tmp/pdfdiff/96a4d6308b5290b6e2df51186d5219c8b9352022/58.png" miff:- ;  convert "/tmp/pdfdiff/edbb34bfe69a279a577b4f9071eb005e6b8f84b6/58.png" miff:- ) | convert - \( -clone 0-1 -compose darken -composite \)  -font helvetica -pointsize 30 -draw "text 10,30 '"'"'Layer: User.9'"'"'"  -channel RGB -combine "/workspace/breakout_hdmi/output/diff/diff-580.png"'
DEBUG:kicad-diff.py:Executing: identify -format '%[colorspace]' /workspace/breakout_hdmi/output/diff/diff-580.png
INFO:kicad-diff.py:Joining all diffs into one PDF
DEBUG:kicad-diff.py:['convert', '/workspace/breakout_hdmi/output/diff/diff-00.png', '/workspace/breakout_hdmi/output/diff/diff-10.png', '/workspace/breakout_hdmi/output/diff/diff-20.png', '/workspace/breakout_hdmi/output/diff/diff-310.png', '/workspace/breakout_hdmi/output/diff/diff-360.png', '/workspace/breakout_hdmi/output/diff/diff-370.png', '/workspace/breakout_hdmi/output/diff/diff-380.png', '/workspace/breakout_hdmi/output/diff/diff-390.png', '/workspace/breakout_hdmi/output/diff/diff-490.png', '/workspace/breakout_hdmi/output/diff/breakout_hdmi-diff_pcb.pdf']
 (kibot - kiplot.py:149)
DEBUG:Removing temporal checkout at /tmp/tmp50ufy24s (kibot - out_diff.py:427)
DEBUG:Executing: /usr/bin/git worktree remove --force /tmp/tmp50ufy24s (kibot - kiplot.py:157)
DEBUG:- CWD: /workspace/breakout_hdmi (kibot - kiplot.py:159)
```
</details>

set-soft, 2023-12-18:
The above patch solves this particular case

closed 2023-12-18 by set-soft

oliv3r, 2023-12-19:
@set-soft thanks! When will this be in the 'latest' docker image so I can try it out? :)

set-soft, 2023-12-20:
Images with KiDiff 2.5.0 are now available.
But I **strongly** suggest you to switch to *dev* images.

commit 72ebff20 (2023-12-18): [Fixed] PNGs not removed

- For pages/layers without difference and using --only_different

Fixes #9