INTI-CMNB/KiDiff #13: Error when running KiDiff in my project

Kampi, 2024-10-31:
I try to run a diff between the `main` and `2.0.6_Dev` branch of [this](https://github.com/kampi/zswatch-hw) repo and get this error:

```
ERROR:Running ['kicad-cli', 'sch', 'export', 'pdf', '-o', '/home/daniel/Test/ZSWatch.pdf', '-t', '_builtin_default', '--black-and-white', '--exclude-drawing-sheet', '--no-background-color', '--pages', '1', '/home/daniel/Test/ZSWatch-HW-main/project/ZSWatch.kicad_sch'] returned -11 (eeschema_do.kiauto.file_util - file_util.py:390)
ERROR:kicad-diff.py:Failed to plot /tmp/tmpj83sg_9r/159ebbac7660aa5e81a391d67290e40f23ea9dc6/Schematic.pdf
```

I´m using KiCad 8.0.6 on Ubuntu 22.04. How can I fix the issue?

set-soft, 2024-11-01:
Please provide the command you used to trigger it

set-soft, 2024-11-01:
Ok, I found the problems:

1. Your page 1 says *#* as page number 
2. KiCad CLI interface crashes if you ask to print page 1, but page 1 can't be found

To solve it just go to "Edit" menu and choose "Edit sheet page number ..." and fix it to be "1"

Kampi, 2024-11-01:
> BTW: Are you using `only_first_sch_page` on KiBot? Because KiBot normally asks for all pages, which won't trigger this issue in KiCad. The problem here is that you don't even have a page **1**

I didn´t use KiBot here. It´s just KiDiff

Kampi, 2024-11-01:
Okay, I changed the page numbers to `1` and reran the command:

```
kicad-diff.py original/project/ZSWatch.kicad_sch modified/project/ZSWatch.kicad_sch
```

Now I get this error:

```
Traceback (most recent call last):
  File "/usr/bin/kicad-diff.py", line 943, in <module>
    output_pdf = DiffImages(old_file_hash, new_file_hash, layers_old, layers_new, args.only_different, changed)
  File "/usr/bin/kicad-diff.py", line 571, in DiffImages
    old = pdf2png(old_file, i not in layers_old, new_file)
  File "/usr/bin/kicad-diff.py", line 422, in pdf2png
    assert False
AssertionError
```

set-soft, 2024-11-01:
--no_reader

Kampi, 2024-11-01:
Got a permission error with ImageMagick. I disabled it in a silly way via 

```
sudo mv /etc/ImageMagick-6/policy.xml /etc/ImageMagick-6/policy.xml.off
```

It´s working now.

Kampi, 2024-11-01:
@set-soft 

can I set a zoom level for a PCB comparison?

closed 2024-11-01 by Kampi

set-soft, 2024-11-01:
Currently uses autoscale for most operations

set-soft, 2024-11-01:
The code uses un-scaled images when the bounding box containing the PCB changes.

Kampi, 2024-11-01:
I changed the resolution to 500 and it looks good now.