INTI-CMNB/KiDiff #14: Output diff is B/W when running in CI/CD

Kampi, 2024-11-03:
Hi,

I want to use CI/CD to generate a diff between a `main` and a `feature` branch. The commands are tested in a Ubuntu VM and there I get colored outputs. The container is outputting B/W.

The workflow is shown below

```
name: Diff

on:
  push:

jobs:
  build:
    runs-on: ubuntu-latest
    container: ghcr.io/inti-cmnb/kicad8_auto_full:latest

    permissions:
      contents: write

    steps:
      - name: Checkout branch
        uses: actions/checkout@v4
        with:
          path: modified
          submodules: recursive

      - name: Checkout main
        uses: actions/checkout@v4
        with:
          ref: main
          path: original
          submodules: recursive

      - name: Create schematics diff
        run: |
          kicad-diff.py original/project/ZSWatch.kicad_sch modified/project/ZSWatch.kicad_sch --all_pages --output_dir diff-schematics --no_reader --keep_pngs

      - name: Create layout diff
        run: |
          kicad-diff.py original/project/ZSWatch.kicad_pcb modified/project/ZSWatch.kicad_pcb --all_pages --output_dir diff-layout --no_reader --keep_pngs

      - name: Upload Layout Diff
        uses: actions/upload-artifact@v4
        with:
          name: "diff-layout-master-${{ github.ref_name }}"
          path: diff-layout
          if-no-files-found: ignore

      - name: Upload Schematics Diff
        uses: actions/upload-artifact@v4
        with:
          name: "diff-schematics-master-${{ github.ref_name }}"
          path: diff-schematics
          if-no-files-found: ignore
```

The output can be viewed [here](https://github.com/Kampi/ZSWatch-HW/actions/runs/11650838162/job/32440053316)

set-soft, 2024-11-04:
Unchanged stuff is always B&W.
This is on purpose, imagine you try to see the diff between 2 F.Cu layers using the classic color scheme: all will be red, giving the impression that all was deleted!
The idea is that you can easily see the difference, only the "2color" mode can configure the colors for the diff.

Kampi, 2024-11-04:
Right - sorry it was a mistake on my side. The branch doesn´t contain any differences.

closed 2024-11-04 by Kampi