INTI-CMNB/KiDiff #16: KiCAD 9 - WARNING:KiCad project file not found on 2.5.7

Tony763, 2025-03-19:
Hello, I trying to use KiDiff on KiCAD 9 with small test project (only one component). I am hitting wall when trying to creating diff of schematic. 

Output from `git diff`:
```
DEBUG:kicad-git-diff.py:['/usr/bin/kicad-diff.py', '--resolution', '150', '--old_file_hash', 'ed2e11bcf38d5f3268e16c424bdc6237fcb0e2ff', '-vv', '--cache_dir', '/home/tony/tmp/test_project/.git/kicad-git-cache', '--exclude', '.kicad-git-diff', '/tmp/git-blob-8acgmR/test_project.kicad_sch', 'test_project.kicad_sch']
DEBUG:kicad-diff.py:/tmp/git-blob-8acgmR/test_project.kicad_sch SHA1 is ed2e11bcf38d5f3268e16c424bdc6237fcb0e2ff
DEBUG:kicad-diff.py:test_project.kicad_sch SHA1 is 383b658e07c3d2a7f22fa3326b32b170b6292018
DEBUG:kicad-diff.py:Cache dir: /home/tony/tmp/test_project/.git/kicad-git-cache
DEBUG:kicad-diff.py:Temporal output dir /tmp/tmp43wf1fgj
DEBUG:kicad-diff.py:layers to be excluded: ['B.Adhes', 'F.Adhes', 'Cmts.User', 'Eco1.User', 'Eco2.User', 'Edge.Cuts', 'Margin']
DEBUG:kicad-diff.py:Cache for /tmp/git-blob-8acgmR/test_project.kicad_sch will be /home/tony/tmp/test_project/.git/kicad-git-cache/ed2e11bcf38d5f3268e16c424bdc6237fcb0e2ff
INFO:kicad-diff.py:cache dir for `/tmp/git-blob-8acgmR/test_project.kicad_sch` already exists
DEBUG:kicad-diff.py:No options for cache entry: /home/tony/tmp/test_project/.git/kicad-git-cache/ed2e11bcf38d5f3268e16c424bdc6237fcb0e2ff/options
INFO:kicad-diff.py:Plotting the schematic
DEBUG:kicad-diff.py:Executing: ['eeschema_do', 'export', '--file_format', 'pdf', '--monochrome', '--no_frame', '--output_name', '/home/tony/tmp/test_project/.git/kicad-git-cache/ed2e11bcf38d5f3268e16c424bdc6237fcb0e2ff/Schematic.pdf', '/tmp/git-blob-8acgmR/test_project.kicad_sch', '.']
WARNING:KiCad project file not found (eeschema_do.kiauto.file_util - file_util.py:317)
Traceback (most recent call last):
  File "/usr/lib/python3.12/shutil.py", line 847, in move
    os.rename(src, real_dst)
FileNotFoundError: [Errno 2] No such file or directory: '/home/tony/tmp/test_project/test_project.pdf' -> '/home/tony/tmp/test_project/.git/kicad-git-cache/ed2e11bcf38d5f3268e16c424bdc6237fcb0e2ff/Schematic.pdf'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/eeschema_do", line 1115, in <module>
    shutil.move(cfg.output_file, cfg.wanted_output_file)
  File "/usr/lib/python3.12/shutil.py", line 867, in move
    copy_function(src, real_dst)
  File "/usr/lib/python3.12/shutil.py", line 475, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/lib/python3.12/shutil.py", line 260, in copyfile
    with open(src, 'rb') as fsrc:
         ^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/home/tony/tmp/test_project/test_project.pdf'
ERROR:kicad-diff.py:Failed to plot /home/tony/tmp/test_project/.git/kicad-git-cache/ed2e11bcf38d5f3268e16c424bdc6237fcb0e2ff/Schematic.pdf
```

Forcing `eeschema_do` showing debug messages with `-vv` points to `No sheets to plot`
```DEBUG:kicad-diff.py:Executing: ['eeschema_do', '-vv', 'export', '--file_format', 'pdf', '--monochrome', '--no_frame', '--output_name', '/home/tony/tmp/test_project/.git/kicad-git-cache/ed2e11bcf38d5f3268e16c424bdc6237fcb0e2ff/Schematic.pdf', '/tmp/git-blob-Ntgl8L/test_project.kicad_sch', '.']
DEBUG:KiAuto v2.3.4 (eeschema_do - misc.py:73)
DEBUG:Detected KiCad v9.0.0 (9.0.0-9.0.0-2~ubuntu24.10.1 9000000) (eeschema_do - misc.py:159)
DEBUG:Config path /home/tony/.config/kicad/9.0 (eeschema_do - misc.py:185)
DEBUG:KiCad environment: {} (eeschema_do - misc.py:305)
DEBUG:KiCad share dir: /usr/share/kicad/ (eeschema_do - misc.py:247)
DEBUG:Defining KICAD9_FOOTPRINT_DIR = /usr/share/kicad/footprints (eeschema_do - misc.py:257)
DEBUG:Setting timeout scale to 1.0 (eeschema_do - eeschema_do:949)
WARNING:KiCad project file not found (eeschema_do.kiauto.file_util - file_util.py:317)
DEBUG:Output file: /home/tony/tmp/test_project/test_project.pdf (eeschema_do - eeschema_do:1017)
DEBUG:Wanted output file: /home/tony/tmp/test_project/.git/kicad-git-cache/ed2e11bcf38d5f3268e16c424bdc6237fcb0e2ff/Schematic.pdf (eeschema_do - eeschema_do:1020)
DEBUG:Executing: kicad-cli sch export pdf -o /home/tony/tmp/test_project/test_project.pdf -t _builtin_default --black-and-white --exclude-drawing-sheet --no-background-color --pages 1 --default-font 'KiCad Font' /tmp/git-blob-Ntgl8L/test_project.kicad_sch (eeschema_do.kiauto.file_util - file_util.py:386)
DEBUG:- Output from command: No sheets to plot.
 (eeschema_do.kiauto.file_util - file_util.py:382)
DEBUG:Changing KiCad name to user selected name /home/tony/tmp/test_project/test_project.pdf -> /home/tony/tmp/test_project/.git/kicad-git-cache/ed2e11bcf38d5f3268e16c424bdc6237fcb0e2ff/Schematic.pdf (eeschema_do - eeschema_do:1113)
```

Think, that there is something wrong with schematics, but running `eeschema_do` on schematics in repository actually work (does not complain about missing project).
 
$ eeschema_do -vv export --file_format pdf --monochrome --no_frame --output_name /home/tony/tmp/test_project/.git/kicad-git-cache/ed2e11bcf38d5f3268e16c424bdc6237fcb0e2ff/Schematic.pdf test_project.kicad_sch .
DEBUG:KiAuto v2.3.4 (eeschema_do - misc.py:73)
DEBUG:Detected KiCad v9.0.0 (9.0.0-9.0.0-2~ubuntu24.10.1 9000000) (eeschema_do - misc.py:159)
DEBUG:Config path /home/tony/.config/kicad/9.0 (eeschema_do - misc.py:185)
DEBUG:KiCad environment: {} (eeschema_do - misc.py:305)
DEBUG:KiCad share dir: /usr/share/kicad/ (eeschema_do - misc.py:247)
DEBUG:Defining KICAD9_FOOTPRINT_DIR = /usr/share/kicad/footprints (eeschema_do - misc.py:257)
DEBUG:Setting timeout scale to 1.0 (eeschema_do - eeschema_do:949)
DEBUG:Output file: /home/tony/tmp/test_project/test_project.pdf (eeschema_do - eeschema_do:1017)
DEBUG:Wanted output file: /home/tony/tmp/test_project/.git/kicad-git-cache/ed2e11bcf38d5f3268e16c424bdc6237fcb0e2ff/Schematic.pdf (eeschema_do - eeschema_do:1020)
DEBUG:Executing: kicad-cli sch export pdf -o /home/tony/tmp/test_project/test_project.pdf -t _builtin_default --black-and-white --exclude-drawing-sheet --no-background-color --pages 1 --default-font 'KiCad Font' /home/tony/tmp/test_project/test_project.kicad_sch (eeschema_do.kiauto.file_util - file_util.py:386)
DEBUG:- Output from command: Plotted to '/home/tony/tmp/test_project/test_project.pdf'.
Done.
 (eeschema_do.kiauto.file_util - file_util.py:382)
DEBUG:Changing KiCad name to user selected name /home/tony/tmp/test_project/test_project.pdf -> /home/tony/tmp/test_project/.git/kicad-git-cache/ed2e11bcf38d5f3268e16c424bdc6237fcb0e2ff/Schematic.pdf (eeschema_do - eeschema_do:1113)
DEBUG:Checking if test_project.kicad_pro was modified (eeschema_do.kiauto.file_util - file_util.py:339)
DEBUG:Checking if test_project.kicad_prl was modified (eeschema_do.kiauto.file_util - file_util.py:339)

set-soft, 2025-03-20:
Without a repo and without a description on how you got it is hard to know, perhaps you are comparing to a faulty schematic, i.e. a 0 bytes files created by KiCad just to have a name there.

> Think, that there is something wrong with schematics, but running eeschema_do on schematics in repository actually work (does not complain about missing project).

But you are comparing to a previous version in the repo, as you don't show the actual command used I can't be sure which one, the faulty one seems to be the previous:

```
DEBUG:kicad-diff.py:Executing: ['eeschema_do', 'export', '--file_format', 'pdf', '--monochrome', '--no_frame', '--output_name', '/home/tony/tmp/test_project/.git/kicad-git-cache/ed2e11bcf38d5f3268e16c424bdc6237fcb0e2ff/Schematic.pdf', '/tmp/git-blob-8acgmR/test_project.kicad_sch', '.']
```

The `/tmp/git-blob-8acgmR/test_project.kicad_sch` is the temporal checkout. And this is the one failing.

Tony763, 2025-03-20:
Test repo there: [test_project.zip](https://github.com/user-attachments/files/19364388/test_project.zip)

set-soft, 2025-03-20:
I tried it and the error from KiCad 9 is annoying but true: you are comparing to an empty schematic.
And some bug in KiCad is preventing from printing it when you ask for page 1.

But I think the best is to modify `kicad-git-diff.py` to include all pages, after all this is what you usually want. Is just that the tool was created before adding the all pages option to `kicad-diff.py`

set-soft, 2025-03-20:
Can you try the above patch?
It will create diffs for all pages.
How did you get the schematic in the repo? Your schematic contains:

```lisp
(kicad_sch (version 20250114) (generator "eeschema") (generator_version "9.0")
  (paper "A4")
  (lib_symbols)
  (symbol_instances)
)
```

But when you create an empty schematic from eeschema you get:

```lisp
(kicad_sch
	(version 20250114)
	(generator "eeschema")
	(generator_version "9.0")
	(uuid "a9c5f9c2-ea30-4d48-a3da-5026ce326f31")
	(paper "A4")
	(lib_symbols)
	(sheet_instances
		(path "/"
			(page "1")
		)
	)
	(embedded_fonts no)
)
```

See the difference? Your schematic doesn't have any page, but an empty one has page number 1. If we ask KiCad to print page number 1 using your schematic it fails because it doesn't have page 1, which sounds ridiculous, but is true.

Tony763, 2025-03-21:
Yes, patched version works, thanks.

Schematic was created during initialization of new project using GUI (KiCAD 9.0.0-2~ubuntu24.10.1)

New project do exact same thing:

[image]

Tony763, 2025-03-22:
Yes, please. I cannot provide full details, thanks.

set-soft, 2025-03-25:
Ok, filled this issue: https://gitlab.com/kicad/code/kicad/-/issues/20448

set-soft, 2025-03-25:
KiCad 9.0.2 won't create bogus schematics.
KiAuto 2.3.5 will catch this situation and create an empty PDF.
I also reported another bug because `kicad-cli` wasn't reporting an error, even when no file was generated (https://gitlab.com/kicad/code/kicad/-/issues/20453)

closed 2025-03-25 by set-soft